adafruit/Adafruit_EPD #2: No good way to disable external SRAM

okofish, 2018-08-17:
[Adafruit_EPD.h](https://github.com/adafruit/Adafruit_EPD/blob/master/Adafruit_EPD.h) provides a `USE_EXTERNAL_SRAM` macro, but `#undef`-ing that after the library is included won't work, as the preprocessor has already gone through the header files. As far as I can tell, to use onboard RAM a user is required to comment out `USE_EXTERNAL_SRAM`, so any projects using this library have to make their own copy :(

Would you accept a PR that adds an optional boolean `useExternalSRAM` argument to the `Adafruit_EPD` constructor? Memory impact should be minimal as the onboard buffers would still only be allocated if `useExternalSRAM` is false.

ladyada, 2018-08-17:
sure we'll take a look! :) ideally you would be able to test if SRAM CS is <0 to know if its connected rather than have a new arg

okofish, 2018-08-17:
Yeah that makes more sense.
One gotcha is that the [hardware SPI/external RAM](https://github.com/adafruit/Adafruit_EPD/blob/master/Adafruit_EPD.h#L82) and the [software SPI/onboard RAM](https://github.com/adafruit/Adafruit_EPD/blob/master/Adafruit_EPD.h#L84) constructors have overlapping argument patterns (e.g. `Adafruit_EPD(int, int, int8_t, int8_t, int8_t, int8_t, int8_t)` is valid syntax for either constructor) so you'd have to remove the onboard RAM one, which would theoretically break existing code that uses it, but this lib is new and everyone's probably using external SRAM anyways so I doubt it'd be a problem.

ladyada, 2018-08-17:
yah i'm ok changing it, or forcing the argument

romainreignier, 2018-08-20:
I have faced the same issue so it is nice to see that someone is working on it.
I was about to change the library using something like this in the sketch before including the library
```C
#define USE_EXTERNAL_RAM 1
// or
#define USE_EXTERNAL_RAM 0
```
And then in `Adafruit_EPD.h`
```C
#ifndef USE_EXTERNAL_RAM
 #define USE_EXTERNAL_RAM 1 // to keep the same default behaviour
#endif
```
And later in the code, replace
```C
#ifdef USE_EXTERNAL_RAM
```
by
```C
#if USE_EXTERNAL_RAM
```
But @okofish solution with an argument is nicer!

ladyada, 2019-03-30:
done now if SRAM_CS is -1

closed 2019-03-30 by ladyada