adafruit/Adafruit_EPD #3: screen flashes black and white every time it's updated

UtCanisAcorio, 2019-01-09:
I need to use the display in a low power mode, but I really only want certain portions of it to change. For example, I want a counter to update periodically. I tried using the epd.print() function, but the example code has the clearbuffer(), fillscreen(EPD_WHITE), and display() functions called every time. I want to update just a portion of the display, i.e., the text position (or for example, a battery gauge icon) that has the value that is changing.  I can't find the source code display() or the other functions, but I definitely don't want the screen to flash black and white multiple times every time I need to update. Is this a requirement for this a screen? I have an e-reader and the display contents update without all that black and white flashing.

ladyada, 2019-01-09:
thats correct, you have to updte the *entire* screen at this time, the displays we stock do not have partial refresh

closed 2019-01-09 by ladyada

UtCanisAcorio, 2019-01-09:
I grabbed a couple of EPDs that just have the SPI interface and nothing else. Is it typical of EPDs to not allow partial refresh?
Also, is it possible to make your display not flash on and off when it refreshes? Can't it just go all white, then load pixels? it's really jarring to see the display go all white then all black back and forth several times.

ladyada, 2019-01-09:
nope!