adafruit/Adafruit_EPD #10: setCursor not working properly.

tavdog, 2019-04-13:
With the Huzzah8266 and EPD Featherwing: calling the two functions one after another but the X position in setcursor is wrong in the first function and correct in the second.
`void display_dir(String s) {
  Serial.println(s);
  epd.setFont(&FreeMonoBold12pt7b);
  epd.setCursor(10, 0);
  epd.setTextColor(EPD_RED);
  epd.print(s);  
}
void display_swell(String s) {
  Serial.println(s);
  epd.setFont(&FreeMonoBold12pt7b);
  epd.setCursor(10, 90);
  epd.setTextColor(EPD_RED);
  epd.print(s); 
}
[image]
`

caternuson, 2019-04-15:
Cross ref:
https://forums.adafruit.com/viewtopic.php?f=57&t=150386

ladyada, 2019-04-15:
@tavdog hmm does it work with the built in font? this would be a bug in GFX not eink i think

tavdog, 2019-04-15:
I narrowed this down to  a leading "\n" in the string.  So if there is a new line the string it brings the cursor back to the edge of the screen I guess.

closed 2019-04-15 by tavdog