adafruit/Adafruit_EPD #11: Issues with drawing in multiple colors on 2.13" E-Ink FeatherWing (Black, Red, White)

pedalPusher68, 2019-04-23:
Hello,

I'm trying to draw graphics on the 2.13" E-Ink FeatherWing that has 3 colors (Black, Red, White).  I'm plugged into the Feather M0 WINC1500 board.  Everything seems to work fine but the images I'm attempting to draw aren't what I'm trying to achieve.

I wrote a simple sketch that draws:

1. A filled round rect - black in color, no overlap with anything else
2. A filled round rect - red in color, no overlap with anything else
3. A filled round rect - black in color
4. A filled round rect - white in color, smaller than and centered on the round rect above (.3)

What's happening is I'm seeing 1., 2., and 3. but I do not see the filled white round rect in the center of the black filled round rect of 3.

Here are relevant lines of code:

`
// Adafruit EPD setup for Feather M0 with 2.13" Feather e-Ink board
#define EPD_CS     9
#define EPD_DC      10
#define SRAM_CS     6
#define EPD_RESET   -1 // can set to -1 and share with microcontroller Reset!
#define EPD_BUSY    -1 // can set to -1 to not use a pin (will wait a fixed delay)

/* This is for using the Adafruit 2.13" tricolor EPD */
Adafruit_IL0373 display(212, 104, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);

void setup() {

    // Set WINC1500 pins for Adafruit MO Feather
    WiFi.setPins(8, 7, 4, 2);

    //Initialize serial and wait for port to open:
    Serial.begin(115200);
    while (!Serial) { ; // wait for serial port to connect. Needed for native USB port only
    }

    sprintf(msg,"Starting Up.");
    Serial.println(msg);

    display.begin();
    Serial.println("display.begin() done...");
    display.clearBuffer();
    display.fillScreen(EPD_WHITE);

    display.fillRoundRect(11,0,40,40,6, EPD_BLACK);
    display.fillRoundRect(74,0,40,40,6, EPD_RED);
    display.fillRoundRect(137,0,40,40,6, EPD_BLACK);
    display.fillRoundRect(147,10,20,20,6, EPD_WHITE);
    display.display();
   . . .
`

There's no other drawing code in this sketch (it's built on the ConnectWithWPA example sketch).  I initially was trying to draw white text strings on a red background on this hardware, but no look so I came up with this overlapping filled rects test.

I'm not sure if I'm doing something wrong - I'd like to not think so, else there needs to be a lot more added to what documentation exists for the EDP library with respect to drawing in multiple colors on these displays.

Any help is appreciated.
Thanks.

ladyada, 2019-04-23:
hiya yeah this is a bug we should be turning off the other layers to make 'white' , we'll take a look but no ETA on fixing quite yet!

pedalPusher68, 2019-04-23:
thanks for the quick response, and, whew!  glad it's not me, I've tried many permutations.  FWIW, I've also had issues getting black to draw over red as well.  Basically any overlapping color drawing operations, as far as I can tell seem to have issues.

Thanks.

ladyada, 2019-04-23:
red ink is thicker than black, so it will always appear 'above' the black layer

ladyada, 2020-12-20:
recently fixed, closing

closed 2020-12-20 by ladyada