adafruit/Adafruit_EPD #15: FeatherWingTest drawBMP setting location not working

caternuson, 2019-12-11:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=57&t=159450

Maybe add the x,y offsets here?
https://github.com/adafruit/Adafruit_EPD/blob/805921829fa7e812eb5ccbd134d99b50fb06050e/examples/FeatherWingTest/FeatherWingTest.ino#L234

ladyada, 2019-12-11:
have them try
https://github.com/adafruit/Adafruit_ImageReader
?

caternuson, 2019-12-16:
@makermelissa does the `FeatherWingTest` example in this repo need updating as well? Or maybe just removed in favor of something in the ImageReader repo?

makermelissa, 2019-12-16:
I think probably removed in favor of the ImageReader repo. I think the EInkBreakout should be able to do feathers as well, but I'd have to double check that.

ladyada, 2019-12-16:
yah plz remove here - imagereader is our canonical image-to-display library!

caternuson, 2019-12-16:
ok @makermelissa can you make that PR? since it sounds like maybe you want to test some things.

makermelissa, 2019-12-16:
Yes, should be easy enough.

makermelissa, 2019-12-17:
Ok, after looking at it, I'm going to make 2 PRs:
1. Add a new EInkFeatherWing test to ImageReader so the breakout one doesn't get messy.
2. Remove any examples in this library that draw bitmaps.

ladyada, 2019-12-17:
sounds like a good plan!

PR #16 (2019-12-17): Removed Bitmap Drawing Examples
This is the second part.

Fixes #15.
closed 2019-12-17 by ladyada