adafruit/Adafruit_EPD #21: When overlaying shapes, red pixels are not overwritten

kDCYorke, 2020-02-19:
May be related to #11 

Code example:

```
#include <Adafruit_GFX.h>
#include <Adafruit_EPD.h>

Adafruit_IL0373 display(152, 152, 1, PIN_A3, 0, -1, -1, &SPI1);

void setup() {
  display.begin();

  display.clearBuffer();
  display.fillScreen(EPD_WHITE);

  // Vertical black bar
  display.fillRect(61, 0, 30, 152, EPD_BLACK);

  // Horizontal red bar overlays
  display.fillRect(0, 61, 152, 30, EPD_RED);

  // Vertical black should overlay but doesn't
  display.fillRect(15, 15, 30, 122, EPD_BLACK);

  display.display();
}

void loop() {
  delay(500);
}
```

When running the above sketch, the smaller black bar shows behind the red bar even though it was added afterwards.

Expected:
[image]

Actual:
[image]

There are two buffers, one for black and one for red. The red gets sent to the screen after the black. The pixel is being set on the color buffer. Then the pixel is set on the black, but it is never cleared on the color. The black buffer is sent, but then the color buffer gets sent and overwrites the pixel.

caternuson, 2020-02-26:
This might be same as #11

kDCYorke, 2020-02-26:
I think it is. I’ve submitted a PR that fixes both.

Kai

ladyada, 2020-12-20:
fixed recently in refactor

closed 2020-12-20 by ladyada