adafruit/Adafruit_EPD #32: Incorrect IL0373 init sequence?

caternuson, 2020-11-10:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=47&t=171658

Something appears to be incorrect with the IL0373 init code definition:
https://github.com/adafruit/Adafruit_EPD/blob/fe9bbed9b8665941ad8d0d597778fd4b7cb28a38/src/drivers/Adafruit_IL0373.cpp#L8-L18

The `IL0373_POWER_ON` line specifies 0 parameters. The next line effectively sends command `0xFF` and specifies `200` parameters, which is beyond the hardwired 64 byte `buf` in `EPD_commandList`. Similar issue near the end for the line `0xFF, 20`. There are not 20 actual parameters specified.

caternuson, 2020-11-10:
Wait, just noticed `0xFF` is a special command:
https://github.com/adafruit/Adafruit_EPD/blob/fe9bbed9b8665941ad8d0d597778fd4b7cb28a38/src/Adafruit_EPD.cpp#L500-L504

:thinking:

caternuson, 2020-11-10:
Don't have hardware to test, but suspecting something with the mix of `uint8_t`, `char` and `int`.

Could `cmd` and `num_args` both be made of type `uint8_t` here?
https://github.com/adafruit/Adafruit_EPD/blob/fe9bbed9b8665941ad8d0d597778fd4b7cb28a38/src/Adafruit_EPD.cpp#L496-L498

anven, 2020-11-10:
I changed the type to uint8_t - behavior did change, but still nor working. Now it gets hung up on "Waiting...."

Adafruit EPD test
set pins
hard reset
busy
done!
	Command: 0x50
	Data: 0x17, 
	Command: 0x82
	Data: 
	Command: 0x2
Flexible config
Text - Buffer clear
Text - Refresh display
  Powering Up
	Command: 0x1
	Data: 0x3, 0x0, 0x2B, 0x2B, 0x9, 
	Command: 0x6
	Data: 0x17, 0x17, 0x17, 
	Command: 0x4
	Data: 
Waiting...

caternuson, 2020-11-10:
Something else may be happening now. But I think the results of that change indicate that the mixed variable types is causing an issue.

**Issue name changed to reflect that.**

PR #34 (2020-11-18): Change some local vars to be more explicit
Fixes #32 

The implicit casting was causing grief on AVR.

Tested on Metro Mini and Itsy Bitsy M4 (to prove it didn't break ARM) using EPDTest.ino.

Here's the Metro Mini setup:
[image]
closed 2020-11-18 by caternuson