adafruit/Adafruit_EPD #46: Error in example acep_7colorTest using Adafruit_FlashTransport_QSPI()

fsteckel, 2021-03-07:
- Arduino board:  All I checked, Uno, ESP32, Feather M4...
- Arduino IDE version 1.8.13
- Adafruit SPIFlash 3.4.0
- Adafruit EPD 4.3.1

When I try to compile your example for the Color EPD (acep_7colorTest) I get the following error:

C:\Users\*****\AppData\Local\Temp\cckKM4GV.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_acep_7ColorTest.ino.cpp.o.9514':
<artificial>:(.text.startup+0x1bc): undefined reference to `Adafruit_FlashTransport_QSPI::Adafruit_FlashTransport_QSPI()'
collect2.exe: error: ld returned 1 exit status
exit status 1

ladyada, 2021-03-07:
hi please see if you can figure out what changed and update the example with a PR!

cantonfoshan, 2022-06-25:
Any news of what changed and is causing the error?

Thanks.