adafruit/Adafruit_EPD #47: Getting RED instead of BLACK with PID 4229

caternuson, 2021-03-09:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=47&t=175386

Seems to be working, but black is showing up as red.

Using a 2.7" Tri-Color eInk / ePaper Shield (PID 4229):
https://www.adafruit.com/product/4229

ladyada, 2021-03-09:
@makermelissa should probably update the example code to use ThinkInk definitions, and test/verify

makermelissa, 2021-03-09:
Yeah, for sure.

makermelissa, 2021-03-12:
Ok, so it turns out I had the CS and DC pins backwards as well as the SRAM CS pin wrong. After fixing those I am getting something that makes sense, except all white is coming through as red for me, everything else looks fine.

makermelissa, 2021-03-12:
Oh, red and black may be backwards as well. I haven't verified it.

PR #48 (2021-03-13): Fix black/red inversion on 2.7-inch EPD
Fixes #47.
closed 2021-03-13 by ladyada