adafruit/Adafruit_EPD #50: Partial update example

tgikal, 2021-04-14:
Feather M4 Express
2.9" E-Ink Featherwing 
Arduino IDE 1.8.14

Using the ThinkInk_partial.ino example in examples, I get a clear "0000" display, but each partial update results in a block that resembles a QR code.
Every 10 updates, during the full screen refresh, the counter displays properly again, I'm not really sure how to troubleshoot this.

ladyada, 2021-04-14:
yeah that is not surprising, the partial code barely works and it wasnt very fun when it did :(

tgikal, 2021-04-14:
The part that's driving me nuts is I effectively have the same or better hardware as used in this demo video: [https://youtu.be/eipKg2pUQ_8?t=280](https://youtu.be/eipKg2pUQ_8?t=280)
But there is no supporting code for how to make it do that.

ladyada, 2021-04-14:
i think they used gxepd, a different library. good luck!

tgikal, 2021-04-14:
With some modifications, calling the proper pins, importing the correct display class, ect. I was able to get a working counter to display with the PartialUpdateTest in the GxEPD library, so I'm a little less frustrated.

I'm sure the GxEPD library isn't designed to use the buffer memory on the display, so it's using a bunch of ram on the processor.

I guess all I can do is hope somebody with a better understanding of this library than me gets the ambition to update it someday.

Zhang--Ning, 2021-05-14:
Hi @tgikal , I got the extract the same issue you had before (using the same hardware you have). I'm happy that you have figured it out. I've tried to debug it but failed. Do you mind elaborating more details on how you did the modifications? Thanks a lot!

digamesystems, 2021-05-21:
Hi all, 

I'm also having issues getting partial update to work on my board. -- In my case, I'm using the 1.54" monochrome display.

[1.54" Monochrome Display](https://www.adafruit.com/product/4196)

I spent a bit working with Mike over on Adafruit's forum and thought it might be good to reach out here. Here's a link to the discussion we had: 

[eInk Partial Update: Supported Boards?
](https://forums.adafruit.com/viewtopic.php?f=19&t=179413) 

I realize that this is a new feature, but we should be able to get there! Any progress on this? 

Thanks!

John

ladyada, 2021-05-21:
there is no ETA - any PRs are accepted but we have no plans to spend more time on it.

digamesystems, 2021-05-21:
Thanks, @ladyada. 

I appreciate the reply. 

Maybe there is a way to leverage some of the work that @ZinggJM has done: 

https://github.com/ZinggJM/GxEPD

I've used his libraries with Waveshare displays and they work quite well. 

From his GitHub page it looks like there may be an opportunity to extend his library to new displays: 

```
I can and will only support e-paper panels I have!

- promotion panels from suppliers are welcome, to add support to GxEPD
- donation panels from users are welcome, to add support to GxEPD
```

'Might be worth sending him a free display?

;)

Regards,
John

ladyada, 2021-05-21:
its probably already supported - please try that library!

digamesystems, 2021-05-21:
Got it! 

If you select: 

`
GxEPD2_BW<GxEPD2_154, GxEPD2_154::HEIGHT> display(GxEPD2_154(/*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16, /*BUSY=*/ 4)); // GDEP015OC1 no longer available
`

as your mapping, in 'GxEPD2_display_selection.h' it works just fine with my display. 

Thanks for the tip! 

Regards,

John

digamesystems, 2021-05-25:
Got it working and made a little example program for the 1.54" Mono Display. 

https://github.com/digamesystems/eInkPartialUpdate

I'm guessing it should work with other monochrome displays if you select the appropriate display in the header above.

gitdode, 2023-06-19:
Could someone please confirm the following on displays using the SSD1680/SSD1681 drivers, regarding the "Display Update Control 2" command (0x22)?

- 0xf7 ("DISPLAY Mode 1") does a full update
- 0xff ("DISPLAY Mode 2") does a partial update
- there is no "DISPLAY Mode" for the fast update mentioned in the [GoodDisplay datasheet](https://v4.cecdn.yun300.cn/100001_1909185147/GDEY0213B74.pdf) on page 10 (I'm just guessing that this is the producer of these displays)

I'm working on a simple "plain" avr-libc support for these displays and while partial update seems to work somewhat, I would be curious about fast update, to save power and possibly longer lifetime of the display - but it seems the SSD168x drivers don't support it?

markbirss, 2023-07-03:
> Could someone please confirm the following on displays using the SSD1680/SSD1681 drivers, regarding the "Display Update Control 2" command (0x22)?
>
> * 0xf7 ("DISPLAY Mode 1") does a full update
> * [...]

Here

https://github.com/ZinggJM/GxEPD2/blob/283ed6366af70fc11dca76705b78b159288edcb7/src/epd/GxEPD2_213_BN.cpp#L410

gitdode, 2023-07-03:
@markbirss Thanks for the interesting pointer. As there is no mention of fast update mode either I guess it is simply not available and I'm looking at the wrong data sheet or the information provided is wrong.

markbirss, 2023-07-03:
> @markbirss Thanks for the interesting pointer. As there is no mention of fast update mode either I guess it is simply not available and I'm looking at the wrong data sheet or the information provided is wrong.
@gitdode 

Yes, I received a 4inch GoodDisplay sample - https://www.youtube.com/shorts/-o7ax1c39Vo
SDD1677 based

and has some additional functions for fast update in the sample code but i modified GxEPD2 support for it

https://www.good-display.com/product/457.html
```
void EPD_HW_Init_Fast(void)
{
  EPD_W21_RST_0;  // Module reset   
  delay(10);//At least 10ms delay 
  EPD_W21_RST_1;
  delay(10); //At least 10ms delay 
  
  Epaper_READBUSY();   
  EPD_W21_WriteCMD(0x12);  //SWRESET
  Epaper_READBUSY();   

  EPD_W21_WriteCMD(0x18);   
  EPD_W21_WriteDATA(0x80); 
  
  EPD_W21_WriteCMD(0x0C);
  EPD_W21_WriteDATA(0xAE);
  EPD_W21_WriteDATA(0xC7);
  EPD_W21_WriteDATA(0xC3);
  EPD_W21_WriteDATA(0xC0);
  EPD_W21_WriteDATA(0x80);
  
  EPD_W21_WriteCMD(0x01); //Driver output control      
  EPD_W21_WriteDATA((EPD_WIDTH-1)%256);   
  EPD_W21_WriteDATA((EPD_WIDTH-1)/256);
  EPD_W21_WriteDATA(0x02);

  EPD_W21_WriteCMD(0x3C); //BorderWavefrom
  EPD_W21_WriteDATA(0x01);  
  
  EPD_W21_WriteCMD(0x11); //data entry mode       
  EPD_W21_WriteDATA(0x03);

  EPD_W21_WriteCMD(0x44); //set Ram-X address start/end position   
  EPD_W21_WriteDATA(0x00);
  EPD_W21_WriteDATA(0x00);
  EPD_W21_WriteDATA((EPD_HEIGHT-1)%256);    
  EPD_W21_WriteDATA((EPD_HEIGHT-1)/256);

  EPD_W21_WriteCMD(0x45); //set Ram-Y address start/end position          
  EPD_W21_WriteDATA(0x00);
  EPD_W21_WriteDATA(0x00); 
  EPD_W21_WriteDATA((EPD_WIDTH-1)%256);    
  EPD_W21_WriteDATA((EPD_WIDTH-1)/256);

  EPD_W21_WriteCMD(0x4E);   // set RAM x address count to 0;
  EPD_W21_WriteDATA(0x00);
  EPD_W21_WriteDATA(0x00);
  EPD_W21_WriteCMD(0x4F);   // set RAM y address count to 0X199;    
  EPD_W21_WriteDATA(0x00);
  EPD_W21_WriteDATA(0x00);
  Epaper_READBUSY();

  //TEMP (1.5s)
  EPD_W21_WriteCMD(0x1A); 
  EPD_W21_WriteDATA(0x5A);

  EPD_W21_WriteCMD(0x22); 
  EPD_W21_WriteDATA(0x91);
  EPD_W21_WriteCMD(0x20);

  Epaper_READBUSY();  
}
```

gitdode, 2023-07-03:
Great, thanks! I wasn't aware they have sample code: https://www.good-display.com/companyfile/631.html

Strooom, 2025-03-01:
Let me try to explain what is the so-called 'Fast Update Mode' : 

When changing the contents on the display, the built-in display controller is going to apply a series of voltages to each pixel which will move the pigment particles to the front or the back. The pigments are suspended in some kind of fluid, and the speed at which they can move depends on the temperature : the higher the temperature, the less viscous the fluid and the faster the pigments move.
So during an update, the display controller needs to know what is the temperature and will select an update sequence depending on that. This temperature measurement can be done either with an internal temperature sensor, or an external. With command 0x18 you select during the initialization which temperature sensor. Default is external, but most libraries will use internal (0x80).

If you use external temperature sensor, you can write the value for the temperature into the controller, using command 0x1A. How to translate a temperature into a 12-bit value is explained in the datasheet.

Now for the fast update : when reverse engineering the example code, it shows that fast update is simple a standard update, but with  the temperature set to (eg) 100 degrees C. So the display thinks it's very warm and can update faster and that's what it does. If you change the written temperature value in these examples to eg -25 C, you will see the fast update is actually slower io. faster...
The example above : 
EPD_W21_WriteCMD(0x1A); // write temperature 
EPD_W21_WriteDATA(0x5A); // temperature = 90 C

I think this is not really a reliable way of updating the display, because if it's really -25C and you will tell the display it should assume its 100C, the pigments won't have enough time to travel front or back and the update will not be as good as with actual temperature used.

However you could try to always trick it into a little higher than actual temperature, and thus speed it up a little. Unfortunately my measurements show that it only really shortens the update time starting from 100 degrees C..

Strooom, 2025-03-01:
> Could someone please confirm the following on displays using the SSD1680/SSD1681 drivers, regarding the "Display Update Control 2" command (0x22)?
> * 0xf7 ("DISPLAY Mode 1") does a full update
> * 0xff ("DISPLAY Mode 2") does a partial update

If you look in the datasheet for Display Update Control 2, there are indeed two modes mentioned, but they are nowhere else explained. So I asked a supplier and they confirmed that Mode 1 is full update and mode 2 is partial update.

I think (information is hard to get on this topic) that full update will completely wipe the display blank, then black and then write the image onto it. So only 1 image is needed in the RAM (eg 0x24 for SSD1681).

For a partial update, three things are different : 

1. you need to write the base image (basically what's on the display before you change something with partial update) to BOTH RAM locations (0x24 and 0x26 for SSD1681)
2. then you write changes to a reduced area in the first RAM bank. You do this by setting xStart, xEnd, yStart, yEnd to a rectangle which contains the pixels to be updated. Then you write new data to only that area.
3. Then you do an update using Mode 2.

I think what's happening is that in Mode 2, the display controller takes the current pixel value from second RAM bank, compares it with new pixel value from first RAM bank, and applies a voltage pattern that updates those pixels depending on the change needed (black to white, white to black, no change).
Afterwards, the new situation is stored in RAM bank 2, so you can do several partial updates by repeating step 2 and 3

gitdode, 2025-03-01:
Thanks @Strooom for the nice explanation.

I've [implemented "fast update mode"](https://github.com/gitdode/thermidity/blob/2f4ebb5e6deafe52330e12efdf6fb6b2b2b5e5b8/thermidity-avr/eink.c#L108) like in the example code and can confirm that it produces a little "ghosting" at temperatures below ~10°C. To remove it, I'm just doing a full update every 10th display update as recommended in the data sheet.

The ghosting is really negligible, but the savings on power are enormous.

Partial update might save even more power (?) but it is also more advanced.

I'm happy with fast update; the thermometers I have built are all running for about 18 months now, the display looks always fine and the 3 AAA batteries are at about 60%.

Strooom, 2025-03-01:
Thank you, that's useful feedback from field trials. I have all update methods working now on an SSD1681 based display so I'm able to compare their timings. I still have to compare energy consumption, but as I have a Nordic PPK2, that should be easy. I will publish the measurements here.