adafruit/Adafruit_EPD #52: Color Inversion on 2.7in Tricolor Display

Dratelle, 2021-06-24:
Library issue with latest versions. Using:
Adafruit 2.7in Tricolor Eink Display (4098)
Arduino Nano 33 IoT
Arduino IDE 1.8.15
Adafruit GFX Library 1.10.10
Adafruit EPD Library 4.4.1
When using EPD examples on the 2.7in displays, the colors are inverted/shifted. So EPD_Black shows up as Red, EPD_RED shows up as white, and EPD_White shows up as white as well. The newer ThinkInk examples do a slighly different remapping: (EPD_RED is black,  EPD_WHITE is red, EPD_BLACK is Red).
I tried swapping various components, and rechecked the wires multiple times. What did make the problem go away was reverting to an older EPD library. I tested a few library versions and found:
4.4.1: Inverted colors
4.3.0: Inverted colors
4.1.0: Inverted colors
4.0.5: Colors correct
4.0.4: Colors correct
4.0.0: Colors correct
3.0.0: Colors correct
So the issue appears to be with 4.1.0 and beyond. Thank you.
[image]

caternuson, 2021-11-08:
Related:
https://forums.adafruit.com/viewtopic.php?f=47&t=184969

makermelissa, 2021-11-10:
Yeah, I kinda know what's going on. I think when the same issue was fixed for the 2.7" Shield (#48), it broke it for the 2.7" breakout. So they're like the reverse of each other. The correct fix should work for both displays, which might be having 2 different initializers or something.

sweig, 2021-12-11:
Ok thanks for the link to #48. This workaround does work for now. A permanent fix by splitting the initializers is a good idea. 

In EPD library 4.4.2 I modified two lines in ThinkInk_270_Tricolor_C44.h:

Line 38/39 NEW (fixed color inversion):
    setBlackBuffer(0, false); //was true
    setColorBuffer(1, false); //was true

But it only works for the ThinkInk examples (#include "Adafruit_EPD.h" using ThinkInk_270_Tricolor_C44 display(...))
It does NOT work for EPD examples (#include "Adafruit_EPD.h" using Adafruit_IL91874 display(...))
For the Adafruit_IL91874 display(...) call I tried a similar mod by changing Adafruit_IL91874.cpp:

Line 151/152:
  setBlackBuffer(0, true); // left as true as false does not work at all
  setColorBuffer(1, false); // was true

Now the EPD examples show black as black. But red now shows as white. Not sure what needs to be changed in addition here.

Why are there two different includes (EPD vs ThinkInk) for the same display in the first place?
Are they being maintained in parallel?

ladyada, 2021-12-11:
EPD was the old library, which still works but was insufficient to describe 'panels' - which had different layers and properties than just the raw chips. EPD is the lower layer, we recommend folks use the ThinkInk layer!

makermelissa, 2022-02-10:
Ok, so I set the Busy and Reset pins both to -1 (to match the shield config) and suddenly the inverted one looks fine.

makermelissa, 2022-02-10:
Ok, it seems the reset pin is the culprit. Busy pin setting has no effect and I've ruled that out. This is kind of strange.

makermelissa, 2022-02-11:
So in trying to find the root cause, I've been taking a look at older versions of code and how it worked and here's some things I found:
- Using EPDTest and Version 3.0.0, was the demo/version combo that I could get working the same on both display setups. 
- With the latest version, the text is red instead of black, but they are at least the same. 
- When using ThinkInk is where it seems to have the reset pin problem, so this is at least something I can compare to figure out more specifically where it's diverging.

makermelissa, 2022-02-11:
Ok, I'm making progress. It has to do with the init codes and buffer settings for the display.

PR #59 (2022-02-11): Fix red inversion regardless of reset pin value
Fixes #52. This fixes a weird error where when the reset pin is passed as anything other than -1, the colors invert.
closed 2022-02-11 by makermelissa