adafruit/Adafruit_EPD #53: Fully convert to BusIO?

caternuson, 2021-08-11:
This library is sort of half way there? `Adafruit_EPD` is using BusIO:
https://github.com/adafruit/Adafruit_EPD/blob/f711ec80ca6eb1b3dc170bc4f8e03c287638ec25/src/Adafruit_EPD.h#L29
but not `Adafruit_MCPSRAM`:
https://github.com/adafruit/Adafruit_EPD/blob/f711ec80ca6eb1b3dc170bc4f8e03c287638ec25/src/Adafruit_MCPSRAM.h#L2

Also, `Adafruit BusIO` is not currently listed as a dependency in `library.properties`.

ladyada, 2021-08-13:
sure, not priority but can do. also busio is a subdep of other deps, if it passes CI, the dep list is fine :)

caternuson, 2021-08-13:
Ah, OK. Chained deps. This'll bring in busio:
https://github.com/adafruit/Adafruit-GFX-Library/blob/f542ec270f264e2aff9a4553172a730b07e559df/library.properties#L10