adafruit/Adafruit_EPD #55: Residual border on MagTag

caternuson, 2021-11-18:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=22&t=185293

CircuitPython usage does not have this behavior. The [Learn guide Arduino example](https://learn.adafruit.com/adafruit-magtag/quotes-example) shows the border as well.

Can recreate same behavior with this example:
```cpp
#include "Adafruit_ThinkInk.h"
#include <Fonts/FreeSans9pt7b.h>

ThinkInk_290_Grayscale4_T5 display(EPD_DC, EPD_RESET, EPD_CS, -1, -1);

void setup() {
  Serial.begin(9600);
  while (!Serial) delay(10);
  Serial.println("Adafruit EPD full update test in mono & grayscale");

  display.begin(THINKINK_GRAYSCALE4);
  display.clearBuffer();
  display.setFont(&FreeSans9pt7b);
  display.setTextSize(1);
  display.setTextColor(EPD_BLACK);
  display.setCursor(10, 30);
  display.print("HELLO WORLD");
  display.display();
}

void loop() {
}
```

ladyada, 2021-11-18:
this would be in one of the init config bytes. its not documented so would have to sorta compare the bytes

caternuson, 2021-11-18:
I tried using the CP config, since there are some differences:
https://github.com/adafruit/circuitpython/blob/932131b4ff4b95066a872b5b299a84e80b7c45d3/ports/espressif/boards/adafruit_magtag_2.9_grayscale/board.c#L45-L52
replacing this:
https://github.com/adafruit/Adafruit_EPD/blob/d55989c0b064c541319fe65e2a663767fe166c96/src/panels/ThinkInk_290_Grayscale4_T5.h#L9-L19
with:
```cpp
static const uint8_t ti_290t5_gray4_init_code[] {
  0x01, 5, 0x03, 0x00, 0x2b, 0x2b, 0x13,
  0x06, 3, 0x17, 0x17, 0x17,
  0x04, 0,
  0xFF, 200,
  0x00, 1, 0x7f,
  0x50, 1, 0x97,
  0x30, 1, 0x3C,
  0x61, 3, 0x80, 0x01, 0x28,
  0x82, 1, 0x12,
  0xFF, 50,
  0xFE // EOM
};
```

But no change. Border is still there.

caternuson, 2021-11-18:
I was wrong about CP not doing this. It does. You can see it here from the [weather guide](https://learn.adafruit.com/magtag-weather):

[image]

And I've recreated that again locally to verify.

As mentioned in the forum thread, this is happening **outside** the addressable display area.

caternuson, 2021-11-18:
This appears to be a feature of the display. There is a border data selection, but unfortunately it is not documented beyond specifying which bits they are: 
[image]

Testing all combos resulted in:
```
VBD[1:0]:
00b dark gray
01b dark gray
10b black
11b light gray (default)
```

So it appears there is no way to turn the border "off".

Closing this for now. Can reopen if more info becomes available on how to set it differently.

closed 2021-11-18 by caternuson

OneOfTheInfiniteMonkeys, 2023-02-16:
To help with some linked items....
https://forums.adafruit.com/viewtopic.php?f=22&t=185293
https://github.com/adafruit/Adafruit_CircuitPython_IL0373/pull/26
https://github.com/adafruit/Adafruit_CircuitPython_IL0373/issues/11

You are not alone.... ;-)
The last post seemt to imply it's fixed, if it is its opaque ( :-o ) to me ;-)

lindes, 2023-08-19:
> To help with some linked items....

Ooh, thanks for those.

> You are not alone.... ;-)

Haha, nope.  :)

> The last post seemt to imply it's fixed, if it is its opaque ( :-o ) to me ;-)

It looks like the fix (which I'd actually chalk up to the pull request of your second link, but that's mentioned in the last link, so fair enough) is in the [Adafruit_CircuitPython_IL0373](https://github.com/adafruit/Adafruit_CircuitPython_IL0373) library... which... I'm guessing the MagTag library uses indirectly?  (I can't see a direct usage of it, and I'm failing to track it down... I think it comes from CircuitPython's board.DISPLAY somehow, but even that I failed to track all the way down... so, perhaps I'm mistaken, but hopefully this is indeed the case.)

Given that (and my assumption that that's what gets used under the covers somewhere), is it possible to expose this new `border` flag from the IL0373 library to be exposed somehow?  If so, could we re-open this issue to track that, or shall I submit a new one, or... ?

Thanks!

OneOfTheInfiniteMonkeys, 2023-08-19:
Hi, well as you will see from the posts in link https://github.com/adafruit/Adafruit_CircuitPython_IL0373/issues/11, dated 09 Aug 2021, things seemed to have happened.  aka comments per Tanweet and Makermelissa.

Everyone seemed to think it could be changed, some even commented it should be changed.

Unfortunately, if there was a change, from my perspective the changes are so opaque, I don't know how to use them.
:-(
Perhaps one day it, will get clearer ;-).

So open source, If I could have figured out or found instructions to compile on a Windows machine, I would have. Then I would not be working on my projects :-(

At the moment I have a Crash Crikey! issue in CircuitPythons >8.0.5 distracting me from anything else.
See here for the current (at time of writing) project https://github.com/OneOfTheInfiniteMonkeys/MTDL where its causing me pain. (I rasied an MQQT % fill display bug issue for the vertical bar displays for that project, it's fixed now - Yeah.)