adafruit/Adafruit_EPD #58: EPD prints only red stuff, after updating the GFX Library from 1.8.5 to 1.10.12

raphael303, 2022-01-21:
I have been using the tri color 2.7" EPD from adafruit in a couple of projects with success. Recently I decided to update the libraries.

My EPD Lib is 4.4.2

After updateing the GFX library to 1.10.12 the EPD display only prints in red, not black. 
Not only in my own code, but also the examples of Adafruit for the EPD do that. For example "EPDTest.ino" 
Everything that is supposed to be black is missing.

I reverted GFX back to 1.8.5 and it works again.

Just wanted to report this as a bug in the library.

ladyada, 2022-01-21:
what if you use the ThinkInk examples?

raphael303, 2022-01-21:
Where can I find them?

jfp5756, 2022-01-21:
On the Arduino IDE, you should find the ThinkInk examples by going to File -> Examples -> Adafruit EPD.

[image]

raphael303, 2022-01-21:
Hi, Thank you, I checked there, but this is how it looks here:
As you can see I am uptodate with Arduino and EPD Library.
[image]

ladyada, 2022-01-21:
mabye you have multiple copies of the libraries - the example are def there
https://github.com/adafruit/Adafruit_EPD/tree/master/examples

makermelissa, 2022-02-11:
Yes, this definitely sounds like multiple versions of the library. If you turn on verbose output for the compiler in preferences, it usually says if it's found multiple libraries and which one it is using. Closing for now because there isn't really anything we can do with the library for this.

closed 2022-02-11 by makermelissa