adafruit/Adafruit_EPD #69: Cannot build epd-test example for STM32 boards

bsatrom, 2022-09-19:
I am trying to run one of the EPD examples on a few different STM32 boards and am not able to build.

- Arduino board:  **STM32F405 Feather**
- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.19**
- List the steps to reproduce the problem below (if possible attach a sketch or copy the sketch code in too):

_I ran the following using the 2.13" 250x122 Tri-Color eInk FeatherWing_

1. Open the `EPDTest` sketch
2. Uncomment line 27 to select the 2.13 Tri-Color (`ThinkInk_213_Tricolor_RW display(EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_BUSY);`)
3. Select the STM32F405 Feather
4. Click Verify/build

```
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.cpp: In member function 'void Adafruit_ACEP::deGhost()':
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.cpp:132:53: error: no matching function for call to 'min(uint32_t&, unsigned int)'
  132 |     uint32_t numbytes = min(remaining, sizeof(block));
      |                                                     ^
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.cpp: In member function 'void Adafruit_UC8151D::displayPartial(uint16_t, uint16_t, uint16_t, uint16_t)':
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.cpp:308:55: error: no matching function for call to 'min(unsigned int, uint32_t&)'
  308 |       uint8_t to_xfer = min(sizeof(mcp_buf), remaining);
      |                                                       ^
In file included from /Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/algorithm:62,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring_constants.h:26,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring.h:34,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/Arduino.h:36,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/Adafruit_MCPSRAM.h:1,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/Adafruit_EPD.h:27,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.h:4,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.cpp:1:
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:3479:5: note: candidate: 'template<class _Tp, class _Compare> constexpr _Tp std::min(std::initializer_list<_Tp>, _Compare)'
 3479 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:3479:5: note:   template argument deduction/substitution failed:
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.cpp:132:53: note:   mismatched types 'std::initializer_list<_Tp>' and 'long unsigned int'
  132 |     uint32_t numbytes = min(remaining, sizeof(block));
      |                                                     ^
In file included from /Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/algorithm:62,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring_constants.h:26,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring.h:34,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/Arduino.h:36,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/Adafruit_MCPSRAM.h:1,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/Adafruit_EPD.h:27,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.h:4,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.cpp:1:
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:3479:5: note: candidate: 'template<class _Tp, class _Compare> constexpr _Tp std::min(std::initializer_list<_Tp>, _Compare)'
 3479 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:3479:5: note:   template argument deduction/substitution failed:
In file included from /Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/algorithm:62,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring_constants.h:26,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring.h:34,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/Arduino.h:36,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/Adafruit_MCPSRAM.h:1,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/Adafruit_EPD.h:27,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.h:4,
                 from /Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.cpp:1:
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:3473:5: note: candidate: 'template<class _Tp> constexpr _Tp std::min(std::initializer_list<_Tp>)'
 3473 |     min(initializer_list<_Tp> __l)
      |     ^~~
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:3473:5: note:   template argument deduction/substitution failed:
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_ACeP.cpp:132:53: note:   mismatched types 'std::initializer_list<_Tp>' and 'long unsigned int'
  132 |     uint32_t numbytes = min(remaining, sizeof(block));
      |                                                     ^
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.cpp:308:55: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned int'
  308 |       uint8_t to_xfer = min(sizeof(mcp_buf), remaining);
      |                                                       ^
In file included from /Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/algorithm:62,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring_constants.h:26,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring.h:34,
[... 51 lines cut ...]
  278 |     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
      |     ^~~
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algobase.h:278:5: note:   template argument deduction/substitution failed:
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.cpp:308:55: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
  308 |       uint8_t to_xfer = min(sizeof(mcp_buf), remaining);
      |                                                       ^
In file included from /Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/algorithm:61,
                 from /Users/satch/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/wiring_constants.h:26,
[... cut ...]
/Users/satch/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/arm-none-eabi/include/c++/10.3.1/bits/stl_algobase.h:230:5: note:   template argument deduction/substitution failed:
/Users/satch/Documents/Arduino/libraries/Adafruit_EPD/src/drivers/Adafruit_UC8151D.cpp:308:55: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
  308 |       uint8_t to_xfer = min(sizeof(mcp_buf), remaining);
      |                                                       ^
exit status 1
Error compiling for board Generic STM32F4 series.

```

bsatrom, 2022-09-19:
BTW it looks like the issue`st_algobase.h` which expects both parameters of it's `min` function to be the exact same type:

```
template<typename _Tp>
    _GLIBCXX14_CONSTEXPR
    inline const _Tp&
    min(const _Tp& __a, const _Tp& __b)
    {
      // concept requirements
      __glibcxx_function_requires(_LessThanComparableConcept<_Tp>)
      //return __b < __a ? __b : __a;
      if (__b < __a)
	return __b;
      return __a;
    }
 ```

performing a cast to int in `Adafruit_ACeP.cpp` and `Adafruit_UC8151D.cpp` results in a successful build.

ClementFoucher, 2022-12-03:
@bsatrom: I see you PR today and it seems that I already proposed a PR to fix this just one day after your issue (which I didn't saw): #68 😄.