adafruit/Adafruit_EPD #77: Line at the top of display

Errorcode95, 2023-09-02:
I am trying to use a WeAct 2.13" B/W Epaper with SSD1680. Everything seams to work fine except there is a line at the top of the display. using Adafruit KB2040 and Arduino IDE 1.8.19.
```
#include "Adafruit_EPD.h"

#define EPD_DC 27
#define EPD_CS 26
#define EPD_BUSY 29
#define EPD_RESET 28
#define EPD_SPI &SPI

Adafruit_SSD1680 display(250, 122, EPD_DC, EPD_RESET, EPD_CS, -1, EPD_BUSY, EPD_SPI);

void setup() {
  Serial.begin(115200);
  display.begin();

  display.clearBuffer();
  display.setCursor(0, 0);
  display.setTextColor(EPD_BLACK);
  display.print("Hello World!");
  display.display();
  delay(5000);
  display.clearBuffer();
}

void loop() {
}
```
[image]

Srafington, 2023-12-05:
You may have an SSD 1675 instead of a 1680. I have a 1675 that came from after Adafruit moved to the 1680 and was trying to treat it like a 1680, and I had the exact issue you are showing here. Soon as I changed to the 1675 class, the line went away

ladyada, 2025-07-08:
you need to set _xram_offset = 0; you can also just use thinkink examples with:
ThinkInk_213_Mono_GDEY0213B74

closed 2025-07-08 by ladyada