adafruit/Adafruit_EPD #85: Missing Online Docs

maxbla, 2024-12-25:
- Arduino board:  N/a

- Arduino IDE version 1.8.19 (but also not N/a)

In the readme of this repository, there is a link to doxygen documentation. Clicking it loads a page that is mostly blank. 
[image]

I checked the [wayback machine for that URL](https://web.archive.org/web/20230715000000*/https://adafruit.github.io/Adafruit_EPD/html/index.html) and it seems to have always been that way. This is strange, since looking through the code, there are comments formatted to work with doxygen. So it seems to be some sort of CI configuration error.

This next part is a bit of a shot in the dark, but I may have spotted the problem. In [an adafruit repository](https://github.com/adafruit/RTClib/tree/master) that has working docs, the [`.github/workflows/githubci.yml`](https://github.com/adafruit/RTClib/blob/master/.github/workflows/githubci.yml) file's last line has an extra argument at the end `run: bash ci/doxy_gen_and_deploy.sh src` whereas [in this repository](https://github.com/adafruit/Adafruit_EPD/blob/master/.github/workflows/githubci.yml) its just `run: bash ci/doxy_gen_and_deploy.sh`

I could submit a pull request to test this idea if a maintainer tells me there's a chance it would be merged

brentru, 2025-08-08:
@ladyada  OK if I fix this while I'm in e-ink land?

ladyada, 2025-08-08:
ya go for it

closed 2026-09-03 by brentru